nRF24/RF24Mesh #233: 2 master slave systems in one room

Chris-656, 2024-02-20:
Is there a possibility to use 2 or even more mesh systems with 1 Master and 3 slaves, so that the 2 different meshes can work without disturbing each other ?
thx Chris

Chris-656, 2024-02-20:
> Not entirely sure what you are asking.
>
> Each mesh network can only have 1 master. I guess the child nodes (not master node) could be considered slaves, but that's not the term we use in the network layers.
>
> Each mesh network can operate on a different channel, but all nodes in a single mesh network have to use the same channel.

When you are using 2 mesh Networks independently in one room.
Do they interferre each other ?

Do the slaves of system 1 only send to master 1.

Regards Chris

2bndy5, 2024-02-20:
Also look at the [`mesh.begin()`](https://nrf24.github.io/RF24Mesh/classESBMesh.html#a44cf6f2b41525b2bc4ba1750364fd8b8). You can easily set the channel for a mesh network using that function. Remember, **each mesh network must have a unique channel**, and all nodes in a single mesh network must use the same channel as the master node.

Chris-656, 2024-02-20:
Thx so it is possible
Regards Chris

closed 2024-02-20 by 2bndy5