nRF24/RF24Mesh #237: Missing node will disable all children.

JonasZaverka, 2024-04-21:
Hi, Im trying to implement this awesome library into my bachelor thesis.
I ran into an issue.
[image]
In this picture all is working fine.
Now I’m going to disable nodID:4 
[image]
Children of nodeID:4 will not get a valid address. This persist until nodeID:4 is back online. 

After nodeID:4 is back online. After some time, all is working fine.
[image]

But for me it’s quite a problem since losing a node is not an unrealistic problem.
Is this issue with my code or setup.
Or perhaps this scenario will not work? 

Thank you, i can send code if anybody suggests that this should work fine but my code is at fault :) 

Thank you very much for your work.

TMRh20, 2024-04-22:
Nodes need to verify their connection periodically. If you look at the examples, you will see exactly this.

I have my nodes checking their connection every 30 seconds, while the example verifies the connection whenever a write fails.

JonasZaverka, 2024-04-22:
if i dont verify the connection nothing happens. After i try sending something it fails and tries to reconnect. That it gets this weird address.  Honestly i ran into memory issues a little later. It may be that. But if i understood correctly.  If one of the nodes goes missing the library should handle reassigning new addresses to all of the missing nodes children. Is that correct ?

2bndy5, 2024-04-22:
>  If one of the nodes goes missing the library should handle reassigning new addresses to all of the missing nodes children. Is that correct ?

The correct answer is no because you have to implement the connectivity handling in user code. See any of the examples and look at where [`renewAddress()`](https://nrf24.github.io/RF24Mesh/classESBMesh.html#a234862c9d9729b3ca11fcb9b29ec55b1) is used.
https://github.com/nRF24/RF24Mesh/blob/7c30debee4b91d63c27847213eaf10d413a10212/examples/RF24Mesh_Example/RF24Mesh_Example.ino#L83-L91

closed 2024-06-21 by 2bndy5