nRF24/RF24Mesh #254: Memory usage on Mesh master device.

mirhamza708, 2024-11-21:
I have an stm32F103CB with 128kB fash and 20kB RAM, I am wondering as to how much of this will be used if I run the mesh master application on this and lets say we have 10 slave nodes connected to this master node. Also will this be possible to connect 100 nodes on this master. Will there be any shortage of RAM and flash?

2bndy5, 2024-11-21:
The DHCP list kept on master nodes allocates 3 bytes per network mode (excluding the master node). With memory alignment (which depends on platform), this can actually be 4 bytes per node (which is actually observed on Linux). So, the allocated memory for just the DHCP list is $`n * a`$ where $`n`$ is the number of nodes in a mesh network (max 255) and $`a`$ is 3 or 4 bytes depending on the platform's memory alignment.

The DHCP list is allocated dynamically though. Upon `RF24Mesh::begin()`, the library allocates enough memory for 10 (`MESH_MEM_ALLOC_SIZE`) nodes. Every time the DHCP list has to grow, memory for 10 more nodes is additionally allocated.

2bndy5, 2024-11-21:
> Will there be any shortage of RAM and flash?

Only if you're using an attiny chip.

> [!tip]
> You can reduce the memory footprint on non-master nodes by defining the macro symbol `MESH_NOMASTER`. This is easily done with platformIO's `build_flags = -DMESH_NOMASTER`, but in Arduino IDE you have to edit this lib's [RF24Mesh_config.h](https://github.com/nRF24/RF24Mesh/blob/7baf4835631c39b3a7f777a7aa16369554c4a70d/RF24Mesh_config.h#L24)

closed 2025-05-07 by 2bndy5