nRF24/RF24Mesh #257: Documentation - Is this accurate?

BHCirrus, 2025-06-03:
I have been reading the documentation prior to implementing a NRF24 network and I have been confused about addressing. I think I know why.

When reading this page: 
https://nrf24.github.io/RF24Mesh/md_docs_2general__usage.html

under Management, the comparison is made that:

"RF24Network addresses can be viewed as MAC addresses, and [RF24Mesh] nodeIDs viewed as static IP addresses".

I think this may be backward because in the Mesh Communications section it is stated:
 
"[RF24Mesh] nodeIDs are unique identifiers, while RF24Network addresses change dynamically within a statically defined structure."

To my understanding of MAC addresses and IP addresses, the nodeID is assigned to the physical device. It is coded into the sketch that is loaded to the device. So this would make it equivalent to the physical address assignment of a MAC address, which is then used to assign a network IP address, which may be dynamic or static depending on the settings of the network. 

Do I have a correct understanding or am I missing something? I am trying to determine the best topography for the network I want to construct so understanding this is critical for my project. 

Thank you!

2bndy5, 2025-06-04:
> To my understanding of MAC addresses and IP addresses, the nodeID is assigned to the physical device. It is coded into the sketch that is loaded to the device. So this would make it equivalent to [...]

This is an pretty good understanding.

I agree the docs are a bit confusing. It is common when writing docs with too much knowledge about implementation on the mind. 

I think of the mesh node ID as the device hostname. In a mesh the hostname doesn't change, but the IP address can. 

<details><summary>This may be more confusing than useful.</summary>

The API is structured like so: 
```mermaid
flowChart
    Bridge["RF24Gateway/Ethernet"]
    RF24 --> RF24Network --> RF24Mesh --> Bridge
    RF24Network --> Bridge
```
Where each API implements different a addressing standard: 
```mermaid
flowChart
    subgraph RF24
    Phy["Physical address"]
    end
    subgraph RF24Network
    Sta["Static addresses"]
    end
    subgraph RF24Mesh
    Meh["dynamically assigned static addresses"]
    end
    RF24 --> RF24Network --> RF24Mesh
```
Physical address (radio pipe address) have a 1-to-1 relationship with static addresses (octal number AKA "logical address" or "network address" in the docs). A mesh node's ID is just an index number used to look up the dynamically assigned static address from master node.

</details>

BHCirrus, 2025-06-04:
Thanks to both of you for the information.

After reading through the replies and the documentation for RF24Mesh and RF24Network, I feel that the documentation would benefit by swapping the comparisons. My reasoning is:

NodeID doesn't change. That is fixed and encoded in the device. This makes it like the MAC address that is *usually* not changed. The device uses the MAC address to request an IP address from the server, just like the mesh client requests a changeable address from the master node. 

So by stating that the NodeID is similar to the MAC address and the network address to an IP address it is, to me, a better fit with common understanding of the two terms. I realize that the IP addresses can match the NodeID, but this is really a convention per my understanding. 

At the end of the day, Node6 may get a different network address every time it gets turned on, but it remains Node6 and there is no other Node6 on the network. So its node ID is really most analogous to a network device MAC address.

Thanks for the time, I understand enough now to implement my system. It just took a lot of extra research because of the way the documentation makes the comparison.

2bndy5, 2025-06-04:
Also checkout [the pyRF24 topology doc](https://pyrf24.readthedocs.io/en/latest/topology.html). It uses the same C++ libraries underneath. Also it outlines the mesh-connect procedure.

BHCirrus, 2025-06-04:
Thanks! I reviewed that before making this issue. It was very helpful and part of what lead me to the conclusion about the nodeID being closer to a MAC address than an IP address.